energia/msp432-core #13: MSP432P401R hangs when Wire.endTransmission() is executed

robertinant, 2016-08-29:
_From @engiecat on February 3, 2016 6:44_

Hello.

I am a user who uses MSP432P401R, with Energia 0101E0017 (and TI CCS Cloud)
While trying to connect a slave i2c device(IMU) to the board, I discovered that MSP432P401R hangs (specifically, goes to 'sleep mode'(according to CCS Cloud debugging) then 'idle mode' when Wire.endTransmission() is executed. )
*Hang means that it doesn't execute next operations, and stuck in the state until it is manually reset.

I had tried several solutions, such as reinstalling Energia, using most recent build of the MSP432's Wire.h library, and even using external pull-up as suggested in http://forum.43oh.com/topic/3660-i2c-wire-library-endtransmission-hang/.

This appears to occur with and without the i2c device connected
(Edit * This I2C device is IMU Digital Combo Board - 6 Degrees of Freedom ITG3200/ADXL345 [SEN-10121])
(In arduino, it doesn't occur even when i2c device is disconnected - just the data is fixed to '0')

Thank you for your assistance in advance.

PS. Just for the reference, I attach the source code i am using which is converted for use in Energia
[FinalVer.txt](https://github.com/energia/Energia/files/115204/FinalVer.txt)

_Copied from original issue: energia/Energia#843_

robertinant, 2016-08-29:
_From @engiecat on February 18, 2016 2:22_

Update : 
It is revealed that this was due to the difference between the operation of Arduino's endTransmission() and Energia's endTransmission. In case of failed communication, while Arduino moves on to the next operation (returning error code), Energia infinitely 'waits' for the signal, resulting in the hang.

This is probably due to the use of the I2C_masterSendMultiByteFinish() instead of I2C_masterSendMultiByteFinishWithTimeout() in the Wire.h library. (IF the library utilizes MSPWare's driverlib.)

robertinant, 2016-08-29:
_From @measley on May 3, 2016 5:33_

I also ran into the same issue using an I2C motor driver. Curious why the library seemed to work on MSP430 but would freeze up on MSP432. After a decent debug session I found root cause to be endTransmission(). Did you find any fix? I can't see where to update wire library.

engiecat, 2016-08-29:
Nope. It seemed to work when i used 1 less endTransmission but it only resulted in garbage data.

robertinant, 2016-10-19:
Further investigation shows that this might now be an Energia issue. If there are no pullups present on the I2C bus (with or without the slave present) Wire.endTransmission() fails. If pullups are applied, the Sketch runs as expected with or without pullups.  @engiecat, I recommend that you apply pullups and you will find that the Sketch runs as expected.

robertinant, 2017-01-10:
I verified this again with the latest release and it works as expected as described in the previous post. Also, when reading from a slave you should follow the following pattern and not call Wire.endTransmission().

```
Wire.requestFrom(2, 6);    // request 6 bytes from slave device #2

while(Wire.available())    // slave may send less than requested
{ 
  char c = Wire.read(); // receive a byte as character
  Serial.print(c);         // print the character
}
```

Below is a Sketch that properly combined reading/writing from/to the slave.

```
#include <Wire.h>

void setup()
{
  Wire.begin(); // join i2c bus (address optional for master)
  Serial.begin(115200);
}

byte x = 0;

void loop()
{
  Wire.beginTransmission(4); // transmit to device #4
  Wire.write("x is ");        // sends five bytes
  Wire.write(x);              // sends one byte  
  Wire.endTransmission();    // stop transmitting

  x++;

  Wire.requestFrom(4, 6);    // request 6 bytes from slave device #4

  while(Wire.available())    // slave may send less than requested
  { 
    char c = Wire.read(); // receive a byte as character
    Serial.print(c);         // print the character
  }
  
  Serial.println(" end  of loop");
  
  delay(500);
}
```

closed 2017-01-10 by robertinant

marsil88, 2017-01-30:
I'm having the same problem with the latest version of Energia and a brand new LaunchPad. It hangs in Wire.endTransmission()

robertinant, 2017-01-30:
@marsil88 can you post your Sketch please? Also, what do you have connected to the I2C bus? Make sure that you have pull-up resistors on the I2C lines.

marsil88, 2017-01-31:
These are the lines regarding I2C:
First I do:
`Wire.begin();`
Then in the first attempt to communicate with the sensor it hangs during endTransmission:

```
Wire.beginTransmission( MS8607_02BA_ADDR_PT ); // the address is 0x76
Wire.write( MS8607_02BA_RESET );// this is the reset command
Wire.endTransmission();// it stops here
```

I do have the pull-up resistors on both lines (4K7 resistors).

hemangjoshi37a, 2017-04-27:
Has anyone found the solution.
I am stuck on this in a big project currently.
Please someone help quickly.
Thank you.

robertinant, 2017-04-27:
As described in the post above. I am only able to reproduce this if no pullups are present.
As mentioned use the pattern as described in this post: https://github.com/energia/msp432-core/issues/13#issuecomment-271687246