energia/msp432-core #32: Sleep() not defined in EMT

robertinant, 2016-08-29:
_From @measley on October 12, 2015 1:41_

Sleep() does not appear to be defined in EMT (MSP432, CC3200)

I replaced sleep() with delay in blink.ino and got following error. 

Blink.ino:32:13: error: 'sleep' was not declared in this scope

I understand the delay already has low power modes enabled, do we need to enable sleep() API for legacy or additional modes?

_Copied from original issue: energia/Energia#742_

robertinant, 2016-08-29:
_From @rei-vilo on December 14, 2015 18:56_

@measley

With EMT, by default, idle mode, *i.e.* no task running = sleep.

robertinant, 2017-01-10:
I could do a sleep macro that simply calls delay.

rei-vilo, 2017-01-10:
> I could do a sleep macro that simply calls delay.

Calling `delay()` would perform a yield.

robertinant, 2017-01-10:
Thinking about this more, I am not going to provide sleep for any EMT targets. MSP430 will remain the only one that will have sleep.

closed 2017-01-10 by robertinant

rei-vilo, 2017-01-10:
As an alternative, for a non-yielding `sleep()`

```
void nonYieldingSleep(uint32_t ms)
{
    uint32_t chrono = millis() + ms;
    while (millis() < chronology);
}
```