energia/msp432-core #33: Implement Watchdog for Energia MT-Supported Boards

robertinant, 2016-08-29:
_From @rei-vilo on December 6, 2015 17:39_

Implement watchdog for the boards supported by EMT.

HAL is provided by `Energia.app/Contents/Resources/Java/hardware/emt/ti/drivers/Watchdog.h`.

Unlike the other RTOS elements proved in the [Galaxia](https://github.com/rei-vilo/GalaxiaLibrary) library, each platform has watchdog-related specifics under the `Energia.app/Contents/Resources/Java/hardware/emt/ti/drivers/watchdog folder`.

* `WatchdogCC26XX.h`
* `WatchdogCC3200.h`
* `WatchdogMSP430.h`
* `WatchdogMSP432.h`
* `WatchdogTiva.h`

_Copied from original issue: energia/Energia#782_

robertinant, 2017-01-10:
I think we should stick this into a library. MSP430 currently (ab)uses the watchdog timer as an interval timer to keep track of millis/micros.

rei-vilo, 2017-01-10:
That would be a great idea.

rei-vilo, 2017-11-11:
See the example and discussion for the CC3200 at http://forum.43oh.com/topic/9346-cc3200-watchdog/

Not sure it works on the MSP432.

rei-vilo, 2017-11-16:
The only caveat is the `WDT_IF_DeInit()` function doesn't seem to deactivate the watchdog.