energia/msp432-core #8: Static code can not configure pins on MSP432

robertinant, 2016-08-29:
_From @dronus on June 18, 2016 16:37_

Even if pure libraries (only using the Arduino / Energia API) are used, some of them show odd behavior on the MSP432.

For example, the LiquidCrystal library only works on certain pin numbers, while crashing the MCU on others. 

It however works well, if it is modified to do all pin initalisation (like pinMode) in begin(), so that it is actually called by setup(). 

I guess the static initalisation is invoked to early, calling pinMode and digitalWrite before device specific initalisation code is ready.

Either it should be discouraged for any library to do static initalisation despite variables, or the init code should be modified to handle this situation, eg. the GPIO subsystem initalised before any setup of the main code is made, including static initalizers.

_Copied from original issue: energia/Energia#901_

robertinant, 2017-01-09:
This is possibly an issue with doing initialization in the libraries constructor. It is discouraged to call any functions in the constructor. There is no guarantee that the system is up at that point. For the msp432, the board initialization functions have not been called yet and hence the call to pinMode() has no affect. The LiquidCrystal needs to be modified to not call pinMode() in the constructor. This should be moved to the LiquidCrystal::begin() function.

dronus, 2017-01-10:
I agree that changing the LiquidCrystal library that way would solve the problem, but as the Arduino platform does not show this problem nor does it seem to define the validity of `pinMode` at certain code locations, it would be very useful to fix this in the MSP library, as this would fix not only LiquidCrystal but maybe a bunch of other libraries suffering from this inconsistency.