energia/msp432-core #38: Two folders contain files with same names but with different content.

rei-vilo, 2017-02-12:
Two folders contain files with same names but with different content. 

* `~/Library/Energia15/packages/energia/hardware/msp432/3.8.0/variants/MSP_EXP432P401R/` (left)
* `~/Library/Energia15/packages/energia/hardware/msp432/3.8.0/cores/msp432/ti/runtime/wiring/msp432/variants/MSP_EXP432P401R/` (right)

[image]

A file in red and green means it is missing on the other side, a file in violet means it is different.

Energia takes the files from both folders. How to prevent any conflict?

robertinant, 2017-02-13:
This should not be there. I just checked the the 3.8.0 package and the only duplicate that I see is the package directory and that should be OK. Did you happen to copy a new package over an older one?

rei-vilo, 2017-02-13:
Update was carried out through the Boards Manager.

rei-vilo, 2017-02-13:
I remove the MSP432 package and then performed a fresh install. Everything goes fine now.

closed 2017-02-13 by rei-vilo