energia/msp432-core #44: Regression Tests on MSP432 non-Red and MSP432 Red

rei-vilo, 2018-03-15:
The example `LCD_LifeGame` for the Kentec 3.5" screen BoosterPack 

+ crashes after 3 rounds when selecting the MSP432 LaunchPad non-Red (board package release 3.8.1), and 
+ fails to run when selecting the MSP432 LaunchPad Red (board package release 5.6.1).

Selecting a small grid (_i.e._ 32 x 24) to avoid memory constraints, and slowing down the SPI speed have no impact.

```
#define ROWS 32
#define COLS 24
```

Three questions:

+ How to check there is no memory leak on the MSP432 LaunchPad non-Red (board package release 3.8.1)
+ For the MSP432 LaunchPad Red (board package release 5.6.1), I suspect the SPI library, as not even the most basic sketch works.
+ Does the SPI implementation differ between MSP432 LaunchPad non-Red and MSP432 LaunchPad Red?

The example used to run fine on previous releases, as I double-checked the example. It still runs fine on the TM4C123 LaunchPad, so the screen isn't the culprit.

robertinant, 2018-03-15:
The MSP432 core is legacy and is no longer maintained. If this bug also happens on the RED board then we can look at it and if the fix applies to the BLACK board then I will apply that fix there as well.

rei-vilo, 2018-05-05:
See https://github.com/energia/msp432r-core/issues/9

closed 2019-05-02 by robertinant